amigavision/AmigaVision #239: Amiga 1200 - missing workbench.library

Skylark13, 2025-01-21:
Hello,

I have the 2024.10.10 release of AmigaVision. After writing games/Amiga/MegaAGS.img (renamed from .hdf) to an SD card with Win32DiskImager, I tried to boot it on my Amiga 1200 (with an ACA 1230 accelerator). It showed a message saying "Please insert a volume containing LIBS/workbench.library in any drive."

[image]

After clicking Cancel on that maybe a dozen times, the title bar color changed to red/pink. After another few times, a shell window appeared showing some text, but then it showed another message saying "Please insert volume WHDSaves in any drive" which I think is #230 .

[image]

After clicking Cancel on that a few times, a workbench screen appeared but no icons or menu functions are available.

[image]

Interestingly, if I load the same SD card as a hard drive in WinUAE (with a config I made very similar to my real A1200 - 68030 with some extra RAM), I only get the WHDSaves error, not the workbench.library one, and I get to a working workbench. So I think the blank workbench I get on the real A1200 is a result of the missing workbench.library which would make sense. But I don't know why I don't get that error in WinUAE, just on the real A1200, given it's booting from the same SD card.

Any idea what is causing the missing workbench.library error on real hardware?

Thanks in advance.

limi, 2025-01-21:
https://amiga.vision/docs#does-amigavision-work-with-kickstart-32

Skylark13, 2025-01-21:
Maybe I'm misunderstanding something, but I think I have 3.1.4? Not 3.2
(granted, I am an Amiga neophyte, so it may well be that I'm missing some knowledge)

[image]

limi, 2025-01-22:
3.1.3 was the last version Commodore made. 3.1.4 and (more commonly) 3.2 were made by Hyperion, an unrelated company with a dubious license to produce software under the name.

It’s confusing, I know. 😄 

Try following the instructions in the FAQ and see if that resolves the issue for 3.1.4 too? I’ll update the entry if it does.

Skylark13, 2025-01-22:
You were right. Thanks for the background info, I didn't know about 3.1.4 nor the reference to CVBA being the dead giveaway! :)

I sent PR #241 as a suggestion for altering the documentation, I think adding the error message makes it more searchable. Feel free to adjust as you see fit of course.

Thanks!

limi, 2025-01-25:
Thanks for confirming!

PR #241 (2025-01-25): Refine 'Kickstart 3.2 / 3.1.4' section
Clarify about 3.1.4 and add specific error message so it will be found when searching.
Fixes #239
closed 2025-01-25 by limi