amigavision/AmigaVision #243: Just checking if any fixes to AmigaVision CD32 games on Mister FPGA

Steveggz, 2025-01-26:
I noticed AmigaVision for Mister was updated in October of 24. I haven't came around to testing yet. Does anyone know if some of the games that were listed as not working are now working?

If I remember correctly Akira, Darkseed, Lost Vikings, Universe, Simon the Sorcerer, D/Generation and Beneath a Steel Sky had issues or weren't working at all.

closed 2025-01-27 by limi

limi, 2025-01-28:
Not that I am aware of. I was hoping that maybe we could get @Slamy (currently working on the CD-i core and also an Amiga programmer) to be interested in tackling CD32 and CDTV cores once the CD-i core is done — especially since most of the work is done with the current Amiga core.

You can follow progress on the CD-i core here: https://github.com/Slamy/CDi_MiSTer

Slamy, 2025-01-28:
@limi I understand your reasoning. We clearly don't have enough core developers on the MiSTer project. As work on the CD-i core is still very hot, I can't say when or if I have time for that. This is a topic I can't currently think about. Too much is still going on and my time is limited. >.<
Sometimes I wish I was a full time MiSTer developer. But I guess something like this doesn't exist.

Steveggz, 2025-01-28:
@limi thank you for the information!

@Slamy thank you for what you do!

I very much wish after the CD-i core is complete you get the urge to look at the CD32 :)

limi, 2025-01-29:
> As work on the CD-i core is still very hot, I can't say when or if I have time for that. This is a topic I can't currently think about. Too much is still going on and my time is limited.

Thanks for responding, @Slamy! Yeah, there simply isn’t enough time to do all the fun things, I agree. Definitely looking forward to seeing the CD-i core progress, and certainly you should take a well-deserved break before even thinking about looking at CD32/CDTV 😄