amigavision/AmigaVision #250: Option to spread Amigavision across 2 .hdf for FAT32 limited MiniMig systems

Higgy69, 2025-02-27:
Hi,

Is it possible to configure Amigavision across 2 mountable .hdf ?
This would be for MiniMig systems that can only work with FAT32 storage but can mount 2 Harddrives (Primary and Secondary IDE) as max single file sizes is 4GB with FAT32.

The first 4GB .hdf could be split 500MB/~3500MB. The standard Amigavision 'Amiga' partition will fit as is. Then on 'Data' it could be WHD folders Demos and Magazines.
Then the second 4GB .hdf would be Games and WHDSaves.

I can't see where AGS2 defines where to look for the G and D folders for example.

I did try hacking about the image to fit on 2 separate booting .hdf 1 for games and 1 for demos but got hunkfile errors (pfs3)  which I've not managed to solve and thus asking on here to more knowledgeable people.

Thanks

limi, 2025-02-28:
We had a version that did this, but decided against it since it would add to the maintenance burden to maintain multiple versions. Now that we have added 4 drive support, we could look at it again, but just as with the Analogue Pocket version that we briefly maintained (because we knew they were working on LBA48 support, which was added), we prefer to have a single setup.

Are there any major setups that only support FAT32 and not exFAT?

limi, 2025-02-28:
The other option would be to revive the “AmigaVision Mini” configuration that we made that slims down the generation of the HDF image to fit within 4GB, while giving you 95% of what people are playing/watching anyway. These are the builds we did for Pocket and (experimentally) the A500 Mini.

Higgy69, 2025-02-28:
Thanks for the reply.

I appreciate this is for a niche within a niche. Turbo Chameleon and FleaOhm for FPGA and all those poor 4GB CFs sat in drawer for use with classic hardware lol!

And I appreciate that it can get messy supporting it. For example my suggestion of putting Games and WHDSaves on a separate 4GB image would make upgrading to newer versions more difficult because WHDSaves would be part of the second .hdf

I will see if I can find one of your older 4GB builds.

Higgy69, 2025-03-01:
I found a pocket image from 2023, probably the last one created. That seems to be working well.
When I try hacking the 9GB onto images I am getting hunkfile errors, I am filling up my 4GB images quite full, I see on Pocket image there is what I would say is a lot of free space, is this the reason for leaving empty space, stop filesystem errors?

If you did revive the "mini" you could potentially add quite a bit extra by altering the Partition sizes to give more to 'Data' by trimming a lot of the Apps in 'Amiga' partition. Unless Pocket users have a dock a lot of that stuff will be tricky to use.

I guess the other option for using the 2023 image is to use a bigger Saves image and add new stuff to the Favourites folder.

Thanks

limi, 2025-03-01:
Yes, we also include a bunch of empty PFS HDFs that you can use as a Saves image. They are in the Extras archive.

Will check the state of the Mini version sometime in the future, right now I need to get the first 2025 release ready 😄

limi, 2026-10-14:
There is a 4GB mini image available now:
https://amiga.vision/a500-mini

closed 2026-10-14 by limi