amigavision/AmigaVision #253: Some WHDLoad Games stuck on WHDLoad loading screen

KoulEl, 2025-03-13:
It's been two versions now that some games won't start and stay on WHDLoad launch screen.

For example Banshee won't start (it loads and don't go further than the WHDload screen startup).

This is on MiST. 2023 edition (I kept this HDF on my SD card) has no problem.

limi, 2025-03-14:
Just tested, and Banshee loads works fine here. Maybe something MiST-specific? Maybe look at WHDLoad binary version and the Banshee container version to see if they are different as a start? Does MiST have problems with files >4GB? (E.g. do you use FAT32 or exFAT?)

KoulEl, 2025-03-16:
I think it's a MiST spefic problem. I'm using an exFAT partition indeed. But former AmigaVision versions (those from 2023) did not show this behaviour. Weird.

limi, 2025-03-17:
Maybe narrow it down through the process of elimination? E.g. Try the 2023 version on the latest version by copying it over, try a third-party version on the image, try a third-party version and the latest version on a simple WB install?

Sorry I can’t do more to troubleshoot MiST, as I don’t have access to the hardware.

KoulEl, 2025-03-17:
Thanks, I'll make some tests and come back to you with specific elements.

KoulEl, 2025-06-05:
New MiST core and firmware fixed these bugs.

closed 2025-06-05 by KoulEl

limi, 2025-06-06:
Great! Thanks for closing the loop.