amigavision/AmigaVision #259: Alien Swarm SE Corrupted Graphics

sh00tthec0re, 2025-03-31:
After walking around the levels for a while, the graphics slowly get more and more corrupted.  I'm running AmigaVision 2024.10.10 on an Amiga 4000.

[image]
[image]

limi, 2025-03-31:
That’s probably something that has to be reported to the WHDLoad team, maybe it hasn’t been properly tested on an 040? I have certainly not seen this on the MiSTer (which is an 020 that is roughly as fast as an 030 at 50MHz).

http://whdload.de/mantis/

limi, 2025-03-31:
If you have the time to try it on a MiSTer (I think you have one, right?) to verify, I’d appreciate it 😊

sh00tthec0re, 2025-04-04:
I did some testing and confirmed that the graphics glitching does not happen on the MiSTer and it also doesn't happen on my Amiga 4000 if I disable my TerribleFire 68060 CPU accelerator and revert to the 68030 stock CPU.  Definitely not a problem you need to worry about - I'll reach out to the TerribleFire folks.

limi, 2025-04-05:
Great! Thanks for the additional testing!

closed 2025-04-05 by limi