amigavision/AmigaVision #261: Game - Stunt Car Racer

dekkit, 2025-04-06:
I'm using MisterFPGA, the Minimig core, and amiga vision.   I noticed a strange issue where the screen resolutions are different for the game Stunt Car Racer depending on how it is loaded.   

Amiga Vision provides both the hdd / A1200 setup with amigavision hdds and it also includes a basic A500 config to enable loading games / demos via .adf disk images.   

I've setup my mister as per the instructions and also checked my settings to ensure the video settings are identical for both setups (as per https://amiga.vision/docs#mister-video-modes).  Both video settings have been set to PAL

I'm using a 5:4 LCD screen connected via HDMI.

When i play the game using an .adf file via the A500 , when actually racing (not the menu screens) the game takes up considerably less space on the LCD screen (e.g. 1/3 less with large black borders).  

When i play the amigavision version of the game, the game takes up more of the game screen and seems looks similar to most videos you see on the web e,g. https://www.youtube.com/watch?v=KoJatWZhcGA.

Any idea on why the Amiga500 setup provided with Amiga vision is using a different resolution? Is there is a setting i need to use to get it looking the same?

Mister Output being reported:

Amiga500
**1496x270** @ 15.63KHz 49.9Hz
1280x1024 @ 108.00MHz 60.0Hz

Amiga Vision
**1280x216** @15.63KHz 49.9hz
1280x1024 @ 108.00MHz 60.0Hz

limi, 2025-04-07:
Yes, this is the 5×PAL scaling that we implement in AmigaVision. If you use the A500 setup, it just uses default PAL. More detail here:

https://amiga.vision/5x/

closed 2025-04-07 by limi

limi, 2025-04-07:
(also, you should add a separate 50hz definition under PAL for your 5:4 screen, you are currently running 50hz games in 60hz, which will cause stuttering)

dekkit, 2025-04-27:
For anyone else using 5:4 LCD Monitor ... copy and paste of my mister.ini

video_mode_pal
video_mode_ntsc

Didn't seem to take effect when toggling between pal/ntsc

However his worked for me to get a nice output at 1280x1024@ 50hz  (thanks for the suggestion)

[Amiga
+Amiga500
+Amiga500HD
+Amiga600HD]

;output in 50hz  (to align with amiga for 5:4 screens - using a custom mode as there wasn't a preset option)
video_mode=1280,1024,50

;Commented out the below
;video_mode_ntsc=8 ; These two use the recommended setting of 1080p60 and
;video_mode_pal=9  ; 1080p50, adjust if you want a different resolution
;vscale_mode=1
;vsync_adjust=1 ; You can set this to 2 if your display can handle it
;custom_aspect_ratio_1=40:27

bootscreen=0