amigavision/AmigaVision #263: Game title added to Favourites but not appearing in the list (MiSTer)

aphexian, 2025-04-18:
I've come across an issue after adding a game to Favourites which then does not appear in the Favourites list. The game is Sensible World of Soccer 96/97. The system definitely thinks the game has been added because if I press Tab on the game again in the main games list I get a message saying that the game has already been added to Favourites, however it still doesn't appear there for some reason. I'm using the current version of AmigaVision from archive.org, AmigaVision-2024.10.10 and I'm using it on an up-to-date MiSTer.

Other games that I've added are appearing in the Favourites list so I'm not sure what the issue with SWOS 96/97 is. I'm not sure if this issue affects any other games.

aphexian, 2025-04-18:
Apologies, just noticed that this issue has already been raised here:

https://github.com/amigavision/AmigaVision/issues/196

Will close this as duplicate.

aphexian, 2025-04-18:
See https://github.com/amigavision/AmigaVision/issues/196

limi, 2025-04-19:
👍

closed 2025-04-19 by limi