amigavision/AmigaVision #264: Crash on Cannon Fodder (MiSTer)

aphexian, 2025-04-18:
I'm getting the following crash on Cannon Fodder immediately after completing Mission 5 Phase 1. This is on the current version of AmigaVision from archive.org, played on a MiSTer.

[image]

limi, 2025-04-19:
Would you mind opening an issue on the WHDLoad site? http://whdload.de/mantis — the issue tracker system is not exactly intuitive, so let me know if you need any assistance.

limi, 2025-06-06:
Also, if you are able to click that Core Dump button and collect the following files from the C: directory, that would be helpful for figuring it out:

```
.whdl_dump
.whdl_expmem
.whdl_memory
.whdl_register
```