amigavision/AmigaVision #271: Uridium 2 hangs on splash screen

IainMcNulty, 2025-05-07:
**AmigaVision Version**
2025.05.05

**Game**
Uridium 2

**Issue**
Game hangs on splash screen after WHDLoad menu.

**Steps to Reproduce**
1. Boot up AmigaVision 2025.05.05 on MiSTer.
2. Select Uridium 2
3. Wait for WHDLoad menu to pass.

**Result**
The title will hang on the following splash screen:

[image]

On previous versions of AmigaVision this title would load without issue.

rsn8887, 2025-05-08:
Works fine for me. Did you press the firebutton to continue after this title screen?

IainMcNulty, 2025-05-08:
I just did a retest and I can close this bug. Pressing fire gets indeed past that splash screen.

However, the user then also has to press fire (again) to get past the next loading screen too (the scrolling rainbow effect).

The user does not have to do this on any disk version of the title at all, nor on any previous version of Uridium 2 on AmigaVision, the title loads automatically without the user having to press fire twice. This new requirement was not documented in the notes for the updated WHDLoad slave for the title either.

Speculation: It was added in the WHDLoad slave so the user didn't miss the splash screen art nor the absolutely amazing loading music due to the title loading quickly otherwise.

closed 2025-05-08 by IainMcNulty